roboticslab-uc3m/Dextra #1: PCB design considerations

jgagom, 2019-02-09:
Ideas to design a PCB layout

jgagom, 2019-02-09:
Regarding PCB traces:

- 45° angles shorten the electrical path between components. 
- High speed logic signals (above 200 MHz) can get reflected off the back of the angle, causing interference.
- 90º has the likelihood of being etched narrower than your standard trace width.
- 90º traces back might not be fully etched, thay could cause shorts.

[image]

andreagil, 2019-02-09:
Steps from breadboard --> to PCB board.

PCB design:
- Schematics design (using KiCad)
- PCB design (routing, components and PCB dimensions)
- BOM list (Bill of materials): actual components to be used (in case connectors of any type have to be considered: USB, expansion jumper connectors...)

SCHEMATICS
It is assumed that each connector is a box with inputs and output. The schematics are the conections between the pins of that boxes without considering the dimensions of any component or the final design of the final PCB (simply the connections, without taking into account the size of hardware).

It is needed:

- **The datasheet** (connections) and **CAD files** (dimensions) from each component (drivers) and the central processor. I assume  drivers are bought (not included in the PCB), but the datasheet is needed to match each output (10 outputs from the photo I have seen). 
[image]

PCB DESIGN
It is needed:

- The schematic.

- **Type of connector**. Doubt: Do I have to take into account connectors? In case the answer is yes, what type of expansion jumper connector are required: SMD (Surface Mounted Device) or through hole connectors?

(Connectors are required in case there are electronic elemens under the drivers which cannot be touching the PCB. These connectors are used as spacers)

So, final doubt: do you want to maintain the black connectors (below image) between the breadboard and the drivers and central processor? or do you want to weld the drivers directly in the PCB holes? (This issue can be considered in the PCB printing process, the process can take into account the black connector or not)

Current black connectors:
[image]

Additionaly, the connections which are going to be ignored are flagged in the next figure
[image]

jgagom, 2019-02-09:
Regarding datasheets: 

- DRV8838 Single Brushed DC Motor Driver Carrier
https://www.pololu.com/product/2990

- Microcontroller: Teensy 3.2
https://www.pjrc.com/store/teensy32.html

- Servo TowerPro MG90 (directly connected to the PCB):
https://servodatabase.com/servo/towerpro/mg90

Regarding CAD files (use right hand ones):

- https://github.com/jgagom/Dextra/tree/dextraTPMG90/Dextra_TPMG90-2_CAD_files

Important [!] - Electronic components won't be welded to the PCB.
Solution: **holes for through-hole components** (check image bellow) for straight 0.1 inch (2.54 mm) spacing (male-male).

[image]

PeterBowman, 2019-06-20:
The design was successfully validated: https://github.com/roboticslab-uc3m/Dextra/issues/5. Can we close this?