roboticslab-uc3m/Dextra #5: Schematics Design

andreagil, 2019-02-12:
In order to design the schematics, only the connections are taken into account.

It is assumed that each connector is a box with inputs and output. The schematics are the conections between the pins of that boxes without considering the dimensions of any component or the final design of the final PCB (simply the connections, without taking into account the size of hardware).

Steps:

1) Create connectors for each driver, encoders and the Teensy following the datasheet information.

2) Add the labels so that KiCad understands each connection between the elements.

jgvictores, 2019-03-27:
Pending: remove via from design

jgvictores, 2019-04-10:
Pending: join PCBs (divide by row to allow cutting, such as holes for a row of pins)

jgvictores, 2019-04-12:
> Pending: join PCBs (divide by row to allow cutting, such as holes for a row of pins)

Perhaps manufacturers can do this for us. We'll find out!

PeterBowman, 2019-06-14:
This design has been validated today on our new right hand PCB. A firmware update is required due to a small change in the Teensy's pinout, see https://github.com/PeterBowman/Dextra/commit/e3cab0e060824f9b8a36afa21fc96b95bcb19211. To be merged soon.

TODO list targeted at future iterations:

* Reverse the Pololu connector pinout, i.e. bring the ground pin back to the left side (so that wires need not to be flipped on their way from the board to the Pololus).
* There are several screw holes, we just need two (top-left and bottom-right). Removing them would leave more room for the CAN transceiver (https://github.com/jgagom/Dextra/issues/2) and the DC/DC converter (https://github.com/jgagom/Dextra/issues/3).

PeterBowman, 2019-06-20:
Merged into default branch at https://github.com/roboticslab-uc3m/Dextra/commit/cc3cdb3ec0158abe4dacd742bfecd29fd41c0f4c. The obsolete right-hand design was removed (still accessible through commit history).

closed 2019-06-20 by PeterBowman